spacelab-ufsc/eps-re #7: Hardware: Add the thermistor to the schematic

joaoclaudioeb, 2023-07-21:
The BQ24650RVAT needs a thermistor to properly adjust the V<sub>MP</sub> of the solar panel when the temperature varies. It is necessary to calculate its value and define its position in the PCB.

joaoclaudioeb, 2023-10-17:
Update: We won't be using a thermistor for now. We're going to use a fixed value of resistors in the "resistive network", on the MPPSET pin. Because we still have to consider how, on the mission, we're going to add this thermistor to the satellite. So, for now, the "resistive network" should be calculated for a fixed value of 25°C.

rebeccaquintino, 2023-10-19:
Update: We used the texas instruments spreadsheet to calculate the resistor network and concluded that 270k ohm is enough to evaluate the tests.

closed 2023-10-20 by joaoclaudioeb