spacelab-ufsc/eps-re #9: Kelvin Connections route and components proximity.

comesanha, 2023-08-26:
From LTC3833 Datasheet:

- Place de sense resistor (R78/R79/R48/R47 in circuit) close to the inductor (L1/L2/L4/L8 - in circuit) on the output side.

- Use a Kelvin (4-wire) connection across the sense resistor (R78/R79/R48/R47 in circuit) and route the traces together as a differential pair. RC filter the differential sense signal close to SENSE+/SENSE– pins, placing the filter capacitor (C18-R43-R44/C19-R45-R46/C106-R64-R65/C107-R62-R61 in circuit) as close as possible to the pins. 

- Place the resistive feedback divider RFB1/2 (R53-R54/R55-R56/R49-R50/R51-R52/R80-R140/R81-R142) as close as possible to VOSNS+/VOSNS– pins and route the remote output and ground traces together as a differential pair and terminate as close to the regulation point as possible. (preferably Kelvin connect across the capacitor at the remote output point)
@joaoclaudioeb

vitoria-b, 2023-09-06:
From the datasheet: "_Place the ceramic CVCC capacitor as close as possible to INTVCC and PGND pins. Likewise, the CB capacitor should be as close as possible to BOOST and SW pins. These capacitors provide the gate charging currents for the power MOSFETs_" should our Cvcc: C110/C111/C22/C23 be closer to 11 and 12 pins or it is in an okay distance? 
[image]

rebeccaquintino, 2023-09-11:
- [X] Place de sense resistor (R78/R79/R48/R47 in circuit) close to the inductor (L1/L2/L4/L8 - in circuit) on the output side.
[image]

- [X] Use a Kelvin (4-wire) connection across the sense resistor (R78/R79/R48/R47 in circuit) and route the traces together as a differential pair. RC filter the differential sense signal close to SENSE+/SENSE– pins, placing the filter capacitor (C18-R43-R44/C19-R45-R46/C106-R64-R65/C107-R62-R61 in circuit) as close as possible to the pins.
[image]
[image]

- [X] Place the resistive feedback divider RFB1/2 (R53-R54/R55-R56/R49-R50/R51-R52/R80-R140/R81-R142) as close as possible to VOSNS+/VOSNS– pins and route the remote output and ground traces together as a differential pair and terminate as close to the regulation point as possible. (preferably Kelvin connect across the capacitor at the remote output point). Done!
[image]
[image]

closed 2023-09-11 by rebeccaquintino